XENONnT/WFSim #39: KeyError: 'No plugin class registered that provides peak_classification'

SidAhmedMa, 2020-01-27:
```python
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-12-0ca034f9b93f> in <module>
      3 
      4 records = st.get_array(run_id,'records')
----> 5 peaks = st.get_array(run_id, ['peaks','peak_classification'])
      6 data = st.get_df(run_id, 'event_info')
      7 

/dali/lgrandi/strax/strax/strax/context.py in get_array(self, run_id, targets, save, max_workers, **kwargs)
    912             results = list(self.get_iter(run_ids[0], targets,
    913                                          save=save, max_workers=max_workers,
--> 914                                          **kwargs))
    915         if len(results):
    916             return np.concatenate(results)

/dali/lgrandi/strax/strax/strax/context.py in get_iter(self, run_id, targets, save, max_workers, time_range, seconds_range, time_within, time_selection, selection_str, **kwargs)
    793         # to merge the results automatically
    794         if isinstance(targets, (list, tuple)) and len(targets) > 1:
--> 795             plugins = self._get_plugins(targets=targets, run_id=run_id)
    796             if len(set(plugins[d].data_kind for d in targets)) == 1:
    797                 temp_name = ('_temp_'

/dali/lgrandi/strax/strax/strax/context.py in _get_plugins(self, targets, run_id)
    413         plugins = collections.defaultdict(get_plugin)
    414         for t in targets:
--> 415             p = get_plugin(t)
    416             # This assignment is actually unnecessary due to defaultdict,
    417             # but just for clarity:

/dali/lgrandi/strax/strax/strax/context.py in get_plugin(d)
    359 
    360             if d not in self._plugin_class_registry:
--> 361                 raise KeyError(f"No plugin class registered that provides {d}")
    362 
    363             p = self._plugin_class_registry[d]()

KeyError: 'No plugin class registered that provides peak_classification'
```

JelleAalbers, 2020-01-27:
Hi Sid, the `peak_classification` datatype was removed in strax 0.4.0. The classification information, i.e. the `type` column, is now integrated in `peak_basics`, see [here](https://straxen.readthedocs.io/en/latest/reference/datastructure.html#peak-basics).

SidAhmedMa, 2020-01-27:
Thanks a lot @JelleAalbers

closed 2020-01-27 by SidAhmedMa

SidAhmedMa, 2020-01-27:
It still not working, I have a new error 
```python
Did not find zx2okm47ra in cache, downloading https://raw.githubusercontent.com/XENONnT/strax_auxiliary_files/master/fax_files/mlp_model.json
---------------------------------------------------------------------------
TimeoutError                              Traceback (most recent call last)
/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/urllib/request.py in do_open(self, http_class, req, **http_conn_args)
   1317                 h.request(req.get_method(), req.selector, req.data, headers,
-> 1318                           encode_chunked=req.has_header('Transfer-encoding'))
   1319             except OSError as err: # timeout error

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/http/client.py in request(self, method, url, body, headers, encode_chunked)
   1238         """Send a complete request to the server."""
-> 1239         self._send_request(method, url, body, headers, encode_chunked)
   1240 

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/http/client.py in _send_request(self, method, url, body, headers, encode_chunked)
   1284             body = _encode(body, 'body')
-> 1285         self.endheaders(body, encode_chunked=encode_chunked)
   1286 

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/http/client.py in endheaders(self, message_body, encode_chunked)
   1233             raise CannotSendHeader()
-> 1234         self._send_output(message_body, encode_chunked=encode_chunked)
   1235 

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/http/client.py in _send_output(self, message_body, encode_chunked)
   1025         del self._buffer[:]
-> 1026         self.send(msg)
   1027 

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/http/client.py in send(self, data)
    963             if self.auto_open:
--> 964                 self.connect()
    965             else:

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/http/client.py in connect(self)
   1391 
-> 1392             super().connect()
   1393 

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/http/client.py in connect(self)
    935         self.sock = self._create_connection(
--> 936             (self.host,self.port), self.timeout, self.source_address)
    937         self.sock.setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/socket.py in create_connection(address, timeout, source_address)
    723     if err is not None:
--> 724         raise err
    725     else:

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/socket.py in create_connection(address, timeout, source_address)
    712                 sock.bind(source_address)
--> 713             sock.connect(sa)
    714             # Break explicitly a reference cycle

TimeoutError: [Errno 110] Connection timed out

During handling of the above exception, another exception occurred:

URLError                                  Traceback (most recent call last)
<ipython-input-13-9884825adf1d> in <module>
      4 records = st.get_array(run_id,'records')
      5 peaks = st.get_array(run_id, ['peaks','peak_basics'])
----> 6 data = st.get_df(run_id, 'event_info')
      7 
      8 truth = st.get_df(run_id, 'truth')

/dali/lgrandi/strax/strax/strax/context.py in get_df(self, run_id, targets, save, max_workers, **kwargs)
    925         df = self.get_array(
    926             run_id, targets,
[... 58 lines cut ...]
    546             return result

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/urllib/request.py in _call_chain(self, chain, kind, meth_name, *args)
    502         for handler in handlers:
    503             func = getattr(handler, meth_name)
--> 504             result = func(*args)
    505             if result is not None:
    506                 return result

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/urllib/request.py in https_open(self, req)
   1359         def https_open(self, req):
   1360             return self.do_open(http.client.HTTPSConnection, req,
-> 1361                 context=self._context, check_hostname=self._check_hostname)
   1362 
   1363         https_request = AbstractHTTPHandler.do_request_

/dali/lgrandi/strax/miniconda3/envs/strax/lib/python3.6/urllib/request.py in do_open(self, http_class, req, **http_conn_args)
   1318                           encode_chunked=req.has_header('Transfer-encoding'))
   1319             except OSError as err: # timeout error
-> 1320                 raise URLError(err)
   1321             r = h.getresponse()
   1322         except:

URLError: <urlopen error [Errno 110] Connection timed out>
```

tunnell, 2020-01-27:
Are you connected to the internet?  Go to https://raw.githubusercontent.com/XENONnT/strax_auxiliary_files/master/fax_files/mlp_model.json with your web browser

zhut19, 2020-01-27:
I also remember having trouble downloading those files when using one of the nodes. Good thing is that this only needs to be downloaded once, so a few ways you can do this:
A) Use straxen.common.get_resource on login node starting like [this](https://xe1t-wiki.lngs.infn.it/doku.php?id=xenon:xenonnt:analysis:guide#setting_up_for_other_things). 
B) Maybe easier to 1) manually download this file from github 2) upload to a /resource_cache folder in your notebook folder but name it zx2okm47ra
C) Ask someone with write access (not me) to put it in that public resource cache folder.

JelleAalbers, 2020-01-27:
I just ran the WFsim tests on the login node, so the file should now be in a public cache folder. Anyone can do this, since we can all write to one of the cache folders strax looks for (`/dali/lgrandi/strax/resource_cache`) and automatically places files in. it's unfortunate the batch jobs don't have access to raw.githubusercontent.com; I think we tried asking but RCC wasn't too keen on it.

SidAhmedMa, 2020-01-28:
I think the problem comes from the fact that I was not working in the right environment, I work in the backup environment and I imagine that in this environment the file zx2okm47ra is not present, which is, by the way still the case. I will try if can add it.