XENONnT/WFSim #44: Update to xenonnt

jorana, 2020-04-24:
As also noted here these values should soon be updated:
https://github.com/XENONnT/strax_auxiliary_files/pull/5

When I've found the correct numbers I'll also update them here:
https://github.com/XENONnT/WFSim/blob/master/wfsim/strax_interface.py#L335

petergaemers, 2020-04-29:
This is fixed with this guy https://github.com/XENONnT/strax_auxiliary_files/commit/458d37fb19c3ae76727ca2d0b4e8dcc35b9e06ca

jorana, 2020-04-29:
Actually this issue was more related to your commit:
https://github.com/XENONnT/WFSim/commit/92a96f62572e0a581caca98361ca7ee371660363

However, people should be aware that this only works if they remove the old file from their cache folders (https://github.com/XENONnT/straxen/blob/365f0f97f6d2a856f14a53e8d0e1649af168d72a/straxen/common.py#L117) . As such, it might be better to version this file in order to avoid many people running into the same problem.

If you do not want to version, it would be nice if people at least don't get the wrong instructions from 'dali/lgrandi/strax/resource_cache'. Did you check this place on dali?

JelleAalbers, 2020-04-29:
Straxen assumes the URLs you give it point to a unique unchanging file, so e.g. writing

``https://github.com/XENONnT/strax_auxiliary_files/blob/dac007c680fc281d82e0c10399e116275d47af5c/fax_config.json``

int he config would work, but

``https://github.com/XENONnT/strax_auxiliary_files/blob/master/fax_config.json``

would give these caching problems. (But I'll admit I haven't always done this properly myself either.)

jorana, 2020-06-24:
Should be fixed in https://github.com/XENONnT/straxen/pull/127.

closed 2020-06-24 by jorana