XENONnT/WFSim #59: Issue Importing WFSim

nikita-0209, 2020-11-13:
If I run these statements:
```
git clone https://github.com/XENONnT/WFSim
pip install -e ./WFSim
```
And then, try this:

```
st = strax.Context(
        storage=strax.DataDirectory('./strax_data'),
        register=wfsim.RawRecordsFromFax1T,
        config=dict(detector='XENON1T',
                    fax_config='https://raw.githubusercontent.com/XENONnT/'
                               'strax_auxiliary_files/master/fax_files/fax_config_1t.json',
                    **straxen.contexts.x1t_common_config,),
        **straxen.contexts.common_opts)
```
I end up with the error:
`NameError: name 'wfsim' is not defined`

zhut19, 2020-11-13:
Hi, once you installed the package successfully, you should able to import wfsim in lower cases, it seems to me, now you are importing by the file path, not the package.

nikita-0209, 2020-11-13:
Hi, running `import wfsim` leads to `ModuleNotFoundError: No module named 'wfsim'`

zhut19, 2020-11-13:
Then I would assume the wfsim is not properly installed, do you get any error during the pip install?

nikita-0209, 2020-11-13:
These are the exact set of statements when I clone and install wfsim

```
Cloning into 'WFSim'...
remote: Enumerating objects: 115, done.
remote: Counting objects: 100% (115/115), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 941 (delta 71), reused 62 (delta 38), pack-reused 826
Receiving objects: 100% (941/941), 3.03 MiB | 5.60 MiB/s, done.
Resolving deltas: 100% (560/560), done.
Obtaining file:///content/WFSim
Collecting strax
  Downloading https://files.pythonhosted.org/packages/b3/8d/a6fe70012871001450b2e25d1ee4ddd93dccf04fb89718673646cd2084cc/strax-0.12.4.tar.gz (123kB)
     |████████████████████████████████| 133kB 13.1MB/s 
Collecting straxen
  Downloading https://files.pythonhosted.org/packages/57/75/bc3bbdec474c0f16771155432e3fec41732ae0ae77d99f9b779fe56a1a3a/straxen-0.12.1.tar.gz (124kB)
     |████████████████████████████████| 133kB 21.0MB/s 
Collecting uproot
  Downloading https://files.pythonhosted.org/packages/e8/3f/642daa0d31868a25fdd4d448ed5105b5d5e71ca25e25bd43e9666df88d07/uproot-3.13.0-py2.py3-none-any.whl (118kB)
     |████████████████████████████████| 122kB 23.6MB/s 
Collecting nestpy
  Downloading https://files.pythonhosted.org/packages/ed/82/d14b68959d7cc98838267601e8ff9bd9716be0d9d1f549754c1b13d4108f/nestpy-1.3.3.tar.gz (76kB)
     |████████████████████████████████| 81kB 8.7MB/s 
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

zhut19, 2020-11-13:
Does the error happen right after the nestpy? I am not able to reproduce this error, but maybe you can try `pip install nestpy` before installing wfsim.

nikita-0209, 2020-11-13:
Okay, yeah I think nestpy leads to the error as `pip install nestpy` leads to
```
Collecting nestpy
  Using cached https://files.pythonhosted.org/packages/ed/82/d14b68959d7cc98838267601e8ff9bd9716be0d9d1f549754c1b13d4108f/nestpy-1.3.3.tar.gz
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```
Perhaps I should put up an issue on the nestpy github repository.

nikita-0209, 2020-11-15:
I installed a stable version of nestpy (1.3.0). And then ran the following commands:
```
!git clone https://github.com/XENONnT/utilix # same here
!pip install -e utilix # install
!git clone https://github.com/AxFoundation/strax.git
!pip install -e ./straxen
!git clone https://github.com/XENONnT/WFSim
!pip install -e ./WFSim
```
And all imports are successful except for:

`ERROR: botocore 1.19.18 has requirement urllib3<1.27,>=1.25.4; python_version != "3.4", but you'll have urllib3 1.24.3 which is incompatible.`

However, `import wfsim` still leads to  `ModuleNotFoundError: No module named 'wfsim'`
I'm working on Google Colabs, do you think there could be a way out?

nikita-0209, 2020-11-18:
It still doesn't work on Google Colab, but I got a local setup and it works there.

closed 2020-11-18 by nikita-0209