XENONnT/WFSim #60: Overlapping in raw_records

HenningSE, 2020-11-17:
I was trying to run WFSim with instructions generated by EPIX. When trying to get the `records` with strax I get the following error: 
`ValueError: Bad data! In channel 165, a pulse starts at -440, BEFORE the previous pulse in that same channel ended (at 0)`
The channel number changes randomly when rerunning the code multiple times. The WFSim version is `0.0.6`.

### How to reproduce the error
First I created the WFSim instructions with EPIX:
```
python3 bin/run_epix.py --InputFile /dali/lgrandi/xenonnt/simulations/nr_materials/mc_v3.0.0/noveto/XENONnT_BellPlate_neutron_50000-001.root --MicroSeparation 0.03 --TagClusterBy energy --Efield /dali/lgrandi/xenonnt/simulations/fieldMapMC.txt --EntryStop 1000  --OutputPath ./
```
You can also skip the EPIX step when using the instructions I already created. They can be found here:
```
/dali/lgrandi/hschulze/instructions_for_wfsim/nVETO_AmBe_neutrons_top_uring_2_wfim_instructions
/dali/lgrandi/hschulze/instructions_for_wfsim/XENONnT_BellPlate_neutron_50000-001_wfim_instructions
```
Next, set the config for WFSim:
```python
st = straxen.contexts.xenonnt_simulation()
run_id = '1'
st.set_config(dict(fax_file=instructions_file))
```
and simulate the `raw_records`:
```python
# Remove any previously simulated data, if such exists
!rm -r strax_data
rr = st.get_array(run_id,'raw_records')
```
With the next line I try to get the `records`:
```python
records = st.get_array(run_id,'records')
```
At this stage I get the error message given above. As additional check I had a look into the `raw_records`as proposed by @WenzDaniel:
```python 
for ch in range(494):
    rc = rr[rr['channel'] == ch]    
    if np.any((strax.endtime(rc)[:-1] - rc['time'][1:]) < 0):
        print(f'Found overlapping records in {ch}') 
```
Overlapping records can be found in multiple channels: 
```
Found overlapping records in 0
Found overlapping records in 1
Found overlapping records in 2
...
```

If I change the WFsim config to disable the checks for overlaps in the `raw records` I can get the `records` without error messages:
```python
st.set_config(dict(check_raw_record_overlaps=False,
                   fax_file= instructions_file,
                   ))
```
When running 
```python
EventInfo = st.get_df(run_id,'event_info')
```
I get this error: `ValueError: Peaks got inconsistent time ranges of inputs: {'peaklets': (-50000, 4999857499), 'merged_s2s': (0, 4999857499)}`

petergaemers, 2020-11-18:
Hi Henning,

Thanks for posting this. I think, but am not entirely sure yet, both issues are on the strax side. The first problem of overlapping records is fixed there https://github.com/XENONnT/straxen/pull/286. For the second I need to have a better look, I suspect it's something in the multithreading code of strax

WenzDaniel, 2020-11-19:
Hi Peter, thanks for your comment, but I strongly disagree with your statement that the "issue" is on the side of strax. There are good reasons for this check, since some of the algorithms may not work properly when having overlapping records. This is simply not to be expected for a real detector. 
I think we should try to work on a solution on the wfsim side to solve this problem. Do you have any suggestion what the best way would be to solve it? Maybe the easiest solution would be an additional plugin which takes the current wfsim output as pre_raw_records and returns raw_records. In general I was wondering if it would not make things easier and the code a bit simpler if we split wfsim among multiple plugins, what do you think?

petergaemers, 2020-11-20:
Hi Daniel, I "strongly suggest" you reread your comment and reconsider. I really have no idea how you manage to have such strong opinions about wfsim while at the same time not understanding what it does. I look forward to seeing your waveform simulator with a 10x speed increase and half the memory consumption. Since you like to suggest everything can be done so much simpler I don't see where I can contribute to this

WenzDaniel, 2020-11-20:
So you disagree, that overlapping pulses may cause issues in straxen?

zhut19, 2020-11-20:
Hi guys, I think this is because WFSim handling unexpected instructions. I test the same instruction file @HenningSE provide and also having trouble producing `records`. But that's because the first instruction has time at 0 and simulation needs padding and push the first raw_records time to -450 and Straxen is not happy with that.
Btw, I believe overlap means endtime - time >= 0, not the other way around.

WenzDaniel, 2020-11-20:
Thanks for clarification we will change the corresponding line and check.

petergaemers, 2020-11-20:
Actually it's not the record timing strax is unhappy about. It is unhappy about the chunk timing, and OverlapWindowPlugins don't understand this. The error comes from here https://github.com/AxFoundation/strax/blob/b77e032632c2951d2e1fddf2eff8d64fb405f3db/strax/plugin.py#L575

We easily avoid this by increasing the time of the first instruction.

closed 2020-11-20 by petergaemers

HenningSE, 2020-11-21:
Thank you guys for the help! It works fine with the first time set to a value larger that zero (e.g. 1e7 nanoseconds)!