XENONnT/WFSim #82: How to implement PMT QE for nVeto

mzks, 2021-02-05:
I create very dirty, broken implementation to add PMT QE for nVeto PMTs. (branch `nveto_pmt_qe`).
The implementation is bad, and it generates errors.

In my naive prediction, only what we have to do is dropping channels and timings after reading Geant4 input, but I may care about ins (instruction) at the same time.
Could you please give me a hint to implement?

For your information,
For nVeto PMTs, I will add static number of QE(*CE) for all PMTs because there are no QE specification sheet for all PMTs. (Half PMTs have measured values.)

WenzDaniel, 2021-02-07:
Hi Keita, also with respect to the time-field issues I mentioned in your PR, I think the best solution here would be to let EPIX handle photon timing and QE for the Veto systems. In this way EPIX would generate two instructions one for the TPC and one for each Veto. In this way we can ensure a correct treatment of all timestamps.

ramirezdiego, 2021-02-08:
Hi Keita, I think I responded here to one of your questions, sorry for the mess: https://github.com/XENONnT/WFSim/pull/81

pietrodigangi, 2021-02-09:
Hi @mzks, just to add that, besides where, I think we should model the nVeto PMT's QE as best as available info allow us.
We have QE spectra for 63 installed PMTs out of 120 in this note: https://xe1t-wiki.lngs.infn.it/doku.php?id=xenon:xenon1t:dsg:muonveto:small_wt:pmts:quantumefficiencybologna
Unfortunatelly Hamamatsu only provided them in their pdf datasheet. with some patience we can extract the spectra from their tables. and randomly assign a QE spectrum to each of the remaining 57 PMTs, taking it from the available ones.

mzks, 2021-02-09:
I've already extracted the QE for 65 available PMTs from pdf.
[image]
The solution which the tables use for 65 PMTs and {random QEs or  average QEs} for the other PMTs, is not beautiful. In addition, the current WFSim doesn't load `pmthitenergy` in Geant4 to calculate them. I think that we can add just an effective constant as a typical QE in this term.

pietrodigangi, 2021-02-10:
wonderful that you already extracted those spectra! :)
I agree that it is not the most beautiful solution, but using a fixed QE is for sure less beautiful given that we have the proper information for half of the PMTs. You can also see that there is quite a variability among those curves, so I think we really want to avoid the constant QE term. Also because we are aiming at detailed nVeto simulations, so we should exploit all the information available as much as possible.

About the energy information, let's discuss it today along with where to actually apply the QE. And then I think we can certainly find the proper way to exploit the pmt hit energy information and apply the QE as a function of it.

mzks, 2021-02-10:
Hi, Pietro
Thank you very much for your comments. I fully agree with you.
In addition, Shingo is asking hamamatsu the QEs for other PMTs. I will start to implement them in WFSim.

ramirezdiego, 2021-02-10:
Reminder of my previous comment: QE values don't go in WFsim, but in the config files repo.

mzks, 2021-03-01:
Done!

closed 2021-03-01 by mzks