XENONnT/WFSim #86: n_top_pmts trated differently

WenzDaniel, 2021-02-09:
The S2-class expects n_top_pmts  to be a number while it is actually an array. This leads to a error in the following line: https://github.com/XENONnT/WFSim/blob/c43e99c875d613db091fb09e519547fb3ffb3b63/wfsim/core.py#L627

zhut19, 2021-02-09:
This config comes from context config via https://github.com/XENONnT/WFSim/blob/d7d399c6eaa7913cac11123e01f2826bac19c805/wfsim/strax_interface.py#L286-L289
I don't see any straxen commit that changes those values to arrays. Are you sure those are arrays in your config?

petergaemers, 2021-02-09:
It should be a number.

WenzDaniel, 2021-02-09:
Okay spotted my mistake I was still initializing my config with the solution Joran proposed. Thanks a lot.

closed 2021-02-09 by WenzDaniel