XENONnT/WFSim #89: New runID argument to communicate with the CMT

ramirezdiego, 2021-02-11:
Ideally, the user would provide a DAQ run ID and we'd query the corrections at that stage, updating the alerady existing CMT interface in WFsim. For the CMT/straxen experts: which complications do you foresee here?

jorana, 2021-02-11:
Actually, this request is better suited in the straxen repo because CMT is not actually a relevant part of WFSim.

>  For the CMT/straxen experts: which complications do you foresee here?
Very little, the most complicated part is most likely having a "political" agreement on:
- Consistent run naming in MC
- What gains to use (what run), when to change this, where to document this
- Actually all using this convention instead of starting a context every time from scratch (i.e. using a common context)

Also tagging @ahiguera-mx because he is the one doing all this CMT stuff.

ramirezdiego, 2021-02-11:
> Actually, this request is better suited in the straxen repo because CMT is not actually a relevant part of WFSim.

Thanks Joran. You're right, but I thought the run id argument would be implemented only in WFsim (since otherwise it is clear that the CMT should read the processed run id, and not a constant value).

About the agreement:
- I think we can diverge in run namings (i.e., add non-integers to MC runs) when the option to call the CMT is implemented in the proposed way.
- This is the main question indeed, but I don't have an answer as I am not familiar with tha gains fluctuation. @ahiguera-mx?
- This is the intention.

ahiguera-mx, 2021-02-11:
I think there is a potential solution for this without modifying CMT, as I understand WFsim want to defined a given `run_id` different from DAQ `run_id` this make total sense, however that would prevent the use of CMT to retrieve actual PMT gains from the corrections DB. To solve this I propose the following approach. 

```
cmt=straxen.CorrectionsManagementServices()

st = strax.Context(
        storage=strax.DataDirectory('/home/'),
        config=dict(detector='XENONnT',
                    fax_config='fax_config_nt.json',
                    check_raw_record_overlaps=False,
                    **straxen.contexts.xnt_common_config,
                    ),
        **straxen.contexts.xnt_common_opts)

# This line will get actual PMT gains from a given run_id: gains_run_id is a run_id from RunsDB
gains = cmt.get_pmt_gains(gains_run_id, 'to_pe_model', 'ONLINE')
straxen.FIXED_TO_PE['WFsim_gains'] = gains

st.set_config(dict(gain_model=('to_pe_constant', 'WFsim_gains')))

st.register(wfsim.RawRecordsFromFaxNT)
```

Then, the only thing is find out how to pass `run_id` to the WFsim context. Can it be part of the fax_config_nt.json? Probably the easies way and we can track any changes to the gains via different versions of fax_config_nt.json

ramirezdiego, 2021-04-22:
Resolved in https://github.com/XENONnT/straxen/pull/443, following the discussion in #108 and here: https://xenonnt.slack.com/archives/C017EMG8QAK/p1618579531066100

closed 2021-04-22 by ramirezdiego