XENONnT/WFSim #114: DPE emission is not applied to S2 maps

terliuk, 2021-03-15:
Current optical maps use QE values provided by Hamamatsu, which are measured by cathode current and, therefore, include DPE emission. In S1 generation, we normalize it to take this effect into account here : https://github.com/XENONnT/WFSim/blob/fac35987b688c8525aac77a86cb74d7fdf0294aa/wfsim/core.py#L331-L332 . 

However, for S2s it doesn't seem to be the case, what will result in g2 overestimated by DPE fraction (QE includes DPE and we add it later again). Here is the corresponding part : https://github.com/XENONnT/WFSim/blob/fac35987b688c8525aac77a86cb74d7fdf0294aa/wfsim/core.py#L542-L543 

Should we simply add the same factor as we used in S1 generation to fix it?

zhut19, 2021-03-15:
You are right @terliuk, the S2 light yield map is not ideal now. The original way of doing this is:

1. Somewhere we need to divide the `np.squeeze(resource.s2_light_yield_map(positions), axis=-1)` by `average s2_light_yield_map(uniform xy positions within max radius)`
  - Which I think is already the case with the [data-driven LCE maps](https://github.com/XENONnT/private_nt_aux_files/tree/master/strax_files)
2. `s2_secondary_sc_gain` should have the unit of n_photon / electron, in contrast to pe / electron. Which in practice we should internally compute this value. i.e. `s2_secondary_sc_gain_photon = s2_secondary_sc_gain_pe / (1 + p_double_photoelectron)`

terliuk, 2021-03-15:
I think the option 2 is the most logical one, as this way we do not change the meaning of our optical maps between S1 and S2 optical maps. 

And a side note:
As for future proofing, we have to keep in mind that actual scaling is `1/mean of PMT response`, since non-amplified photoeletrons also contribute to mean that is provided by Hamamatsu. And, probably, it has to be applied to individual PMTs rather than global value, but we need to have more detailed analysis to check it.

terliuk, 2021-05-01:
@zhut19 Does you PR https://github.com/XENONnT/WFSim/pull/143 also fixes this issue?

ramirezdiego, 2021-05-01:
@terliuk I think this is the line, right?
https://github.com/XENONnT/WFSim/blob/c6ff9848674acde11fb0bf92697039d37f204c79/wfsim/core.py#L586-L588
Compare with the one pasted above on March 15th.

So we can probably close this, but not sure if the side note by @zhut19 deserves another issue, to keep track...

terliuk, 2021-05-01:
Yes, that was the intention, just to check whether I understand it correctly. I think we can close this issue as resolved now.

closed 2021-05-01 by terliuk