XENONnT/WFSim #126: entry start/stop in McChainSimulator

ramirezdiego, 2021-03-25:
What I truly want is to submit review comments to https://github.com/XENONnT/WFSim/pull/103, but this would be a bit messy, since it is already merged, so I'll push here the changes we agree on: https://github.com/XENONnT/WFSim/pull/125

- First comment/request for @petergaemers: could you please resolve the open conversations in PR https://github.com/XENONnT/WFSim/pull/103? I have some comments which I am not sure are repeated.

I want to first revisit the discussion on these arguments, which are always required for running the chain:
https://github.com/XENONnT/WFSim/blob/9619aa7e16f2425553bbded07608d03e9cff00d8/wfsim/strax_interface.py#L233-L234

- For example, in this function we use `event_start` and `event_stop` to calculate the rate
https://github.com/XENONnT/WFSim/blob/9619aa7e16f2425553bbded07608d03e9cff00d8/wfsim/strax_interface.py#L247-L257
If `event_stop` is equal to `-1` (default), the code searches for the maximum G4 event id with this function, and assigns it to `event_stop` (by the way, I don't understand why the two last lines are needed):
https://github.com/XENONnT/WFSim/blob/9619aa7e16f2425553bbded07608d03e9cff00d8/wfsim/strax_interface.py#L208-L214
I think this is not correct, since the last G4 id does not necessarily correspond to the last event simulated, which is the number that we should use to distribute the source rate.

- Related to this, this function is only useful if we envision to decouple single ROOT files into multiple smaller ones, correct? https://github.com/XENONnT/WFSim/blob/9619aa7e16f2425553bbded07608d03e9cff00d8/wfsim/strax_interface.py#L216-L220
This function is redundant with the job epix already does, if you these two arguments always as epix config. Just to be clear: the default rate we want for epix in the `mc_chain` is 0, correct (i.e., no time correction applied in epix)?

petergaemers, 2021-03-25:
Whats the open conversation of 103?

What are the two last lines? the start= blabla  are for readability, els the lines become very long

The last g4 id is the last one we can simulate right? So thats the one to stop after since nothing more is coming

The cut function is redundant as of now. When implementing this, this wasn't in epix.

And yes the default source rate for epix is 0

ramirezdiego, 2021-03-25:
> 
> 
> Whats the open conversation of 103?
> 
All the review comments for which you didn't click on "Resolve Conversation". I wanted to have a close look at the review before pushing the needed changes in mc_chain, but it isn't clear what you accepted/fix and what you didn't. Sorry if I am tedious here.

> What are the two last lines? the start= blabla are for readability, els the lines become very long
> 
https://github.com/XENONnT/WFSim/blob/9619aa7e16f2425553bbded07608d03e9cff00d8/wfsim/strax_interface.py#L213-L214

> The last g4 id is the last one we can simulate right? So thats the one to stop after since nothing more is coming
> 
No, this is why we should change the logic of this id cut. The last g4 id in the epix input file (ROOT) is the last simulated event which produced a hit in a sensitive detector (LXe, PMTs...). This means that we can simulate 50 events and our last g4 id could perfectly be 45. In that case we don't properly distribute the source rate, since we normalize wrt 45.
This is definitely something that we must fix here and/or in the mc_chain input arguments.

> The cut function is redundant as of now. When implementing this, this wasn't in epix.
> 
Indeed, but I had understood that the tests for this branch were done with the strax_interface branch of epix (before it was merged). My review now comes from the fact that I think the integration was not as clean as we planned. I guess that the event cut must remain in epix then, and we delete the function from here?

> And yes the default source rate for epix is 0

Ok.

petergaemers, 2021-03-25:
Everything is adressed.

Oke the last g4 id then is easy, just do event_stop=n_events (or whatever the argument is for number of events to produce in g4)

I mean the cut should be in epix, and this guy will go

ramirezdiego, 2021-03-25:
@petergaemers One more question: why do the `event_rate` units need to be ns^-1?
https://github.com/XENONnT/WFSim/blob/9de9ebad2706a8fd3f2cab4dc9d7c68cdacfa6ca/wfsim/strax_interface.py#L251-L254

ramirezdiego, 2021-03-26:
Resolved in #125 and https://github.com/XENONnT/mc_chain/pull/12.

closed 2021-03-26 by ramirezdiego