XENONnT/WFSim #132: No 1T full-chain compatibility

ramirezdiego, 2021-04-02:
The 1T compatibility was skipped in the full-chain WFSim classes development: https://github.com/XENONnT/WFSim/pull/103
See this class
https://github.com/XENONnT/WFSim/blob/e2fc8b87bd3e4a8fc130b2b081a642576387a6ad/wfsim/strax_interface.py#L675
only called when the nT simulations context is active.

This is not a big deal right now. However, MC and epix are 1T-compatible, and the mc_chain scripts were written so that 1T full-chain simulations can also be ran (maybe a nice option for validation?). In addition, WFSim itself was already used for 1T and is compatible, so only a small refactoring of these classes would be needed to enable 1T full-chain simulations. I cc @petergaemers to comment if the latter statement is incorrect and lots of work would be required here (maybe you think it makes sense that, for 1T, we run epix separately and apply the time shifts directly in there?); and I cc the ACs (@jpienaar13, @kdund, @skazama) in case they want to share their thoughts.

In any case I'm not strongly concerned about this, but just wanted to make people aware.

HenningSE, 2021-04-06:
In case of the XENON1T simulations do we want to process the data only with straxen or do we want to be able to use it with PAX and HAX? Some time ago one could use WFsim with the old software but I think support for this old software is not needed anymore or?

petergaemers, 2021-04-07:
Currently it's just not there. My personal view is if someone wants it they can copy the nt part and make the needed changes

ramirezdiego, 2021-04-16:
Resolved in #125.

closed 2021-04-16 by ramirezdiego