XENONnT/WFSim #134: Quanta from nestpy called twice

sophiaandaloro, 2021-04-06:
I just noticed in this tutorial that the quanta generation is done twice on the yields. The reason why this is an issue is that in one call, the total quanta is conserved, but if you call GetQuanta twice, there's no telling what n_e + n_photons will add up to be and no validity test that they will be equal to the total quanta number. I am unsure if this is replicated throughout the code, in which case it will require updating as soon as possible. 

Here's how it looks now in the [tutorial](https://github.com/XENONnT/WFSim/blob/master/notebooks/Getting_started_wfsim.ipynb):
```
quanta.append(nc.GetQuanta(y, density).photons)
quanta.append(nc.GetQuanta(y, density).electrons)
```

And here is how you want it to look:
```
q = nc.GetQuanta(y, density) # or whatever you want to call it
quanta.append(q.photons)
quanta.append(q.electrons)
```

Let me know if I am missing something or if you need more clarifications! Thanks!

ramirezdiego, 2021-04-06:
Thanks for spotting this, @sophiaandaloro! If this means you're actually running the tutorial, it would be great if you can just push the fix, together with any line or explanation you consider as missing, since the tutorials haven't been updated for a long while...

About the code itself, there's no nestpy call in WFSim since end of last year (https://github.com/XENONnT/WFSim/pull/64), since now epix does this job (precisely [here](https://github.com/XENONnT/epix/blob/master/epix/quanta_generation.py)). However, I just looked into the old WFSim code and this issue was present, so still a good catch! Luckily this entire nestpy+WFSim format wasn't used back then and people mostly coded their own generator.

sophiaandaloro, 2021-04-06:
Ah ok. Well that is great. I don't have this installed, but it should be easy for me to make the update now (assuming the clone goes fine.) Will close this issue with the PR/commit once made.

sophiaandaloro, 2021-04-06:
See #135 for PR.

closed 2021-04-06 by sophiaandaloro