XENONnT/WFSim #139: HISTORY and PyPI

tunnell, 2021-04-13:
If you putting it on PyPI, I'd keep the HISTORY file up to date because otherwise the PyPI page when you are looking for the package only shows really outdated irrelevant info.

tunnell, 2021-04-13:
Also, see this in setup.py:  'Development Status :: 2 - Pre-Alpha'

tunnell, 2021-04-13:
Also, this all started with trying to figure out why WFsim versions were appended by a "0." for a student.

jorana, 2021-04-13:
>, I'd keep the HISTORY 
@tunnell , I agree and made a similar comment/PR:
https://github.com/XENONnT/WFSim/pull/123

ramirezdiego, 2021-04-13:
Agreed as well. No need for an immediate HISTORY update, but a proper one following #125 (and new release) is planned.

jorana, 2021-05-03:
With the recent releases and #150 this issue seems solved

closed 2021-05-03 by jorana