XENONnT/WFSim #144: strax dislikes two different defaults for gain_model_nv

ramirezdiego, 2021-04-30:
When running
```
st.register(wfsim.RawRecordsFromMcChain)
```
strax triggers an error:
```
/opt/XENONnT/anaconda/envs/XENONnT_development/lib/python3.8/site-packages/strax/context.py in register(self, plugin_class)
    280                                    ' one of the defaults.'
    281                                    )
--> 282                             raise ValueError(mes)
    283 
    284                 except strax.InvalidConfiguration:

ValueError: Two plugins have a different default value for the same option.
The option "gain_model_nv" in "muVETOHitlets" takes as a default "('CMT_model', ('to_pe_model_nv', 'ONLINE'))"
while in "RawRecordsFromMcChain" the default value is set to "('to_pe_constant', 0.01)".
Please change one of the defaults.
```

I open the issue here (WFsim) since I guess it can be fixed by changing this guy:
https://github.com/XENONnT/WFSim/blob/f3f5daca71a505604ed8039ba5f0127a639c5669/wfsim/strax_interface.py#L524

... but experts' advice on whether this is the best solution is welcome. I assume there's no default needed for thsi line, but then one needs to always overwrite this model via context (like @petergaemers fixed here: https://github.com/XENONnT/straxen/pull/446), which needs to be clearly documented.

petergaemers, 2021-04-30:
Option used by multiple classes shouldn't be specified in the class but it the context I think

ramirezdiego, 2021-04-30:
Okay, this means removing the default is indeed the way to go?

petergaemers, 2021-04-30:
Yes here and also in muVetoHits

ramirezdiego, 2021-04-30:
@WenzDaniel Do you consider also removing the default here?
https://github.com/XENONnT/straxen/blob/ccc19afc20fb18570a66ab0dcb20535b10fb59e0/straxen/plugins/veto_hitlets.py#L43

ramirezdiego, 2021-04-30:
Fixed here: #145
On the straxen end, see conversation: https://github.com/XENONnT/straxen/pull/460

closed 2021-04-30 by ramirezdiego