XENONnT/WFSim #165: Adding NEST to S1 scintillation models

zhut19, 2021-06-03:
We should include [nestpy ](https://github.com/NESTCollaboration/nestpy) in our S1 scintillation model, as also suggested by @terliuk 's study.

ramirezdiego, 2021-06-03:
Related to this issue, I refer to this message: https://xenonnt.slack.com/archives/C0177G05M1P/p1620812142120200?thread_ts=1620811369.118500&cid=C0177G05M1P
Let's please make clear in the config files which paremeters are used for which model.

terliuk, 2021-06-04:
I made propagation-only timing splines in addition one containing Kr83m delay (format is more or less the same). I will clean up the code that I use to make them in addition to a file with splines. BTW, we decided to go with top/bottom array timing, right? 

Other parts are rather straightforward:
 - passage of local field, number of photons, excitons and deposit energy to S1 class, as this is the information needed by nestpy 
 -  think about the place, where we want to put netspy object, similar to what we have with resources

ramirezdiego, 2021-06-04:
Cool!

> 
>     * passage of local field, number of photons, excitons and deposit energy to S1 class, as this is the information needed by nestpy
> 

This one points directly to epix. We discussed this in the MC meeting where the splines propagation was presented and (I think) agreeed on not duplicating to much epix does already, right? Meaning that these parameters would take care of time-only in WFSim and not quanta generation. Just making sure we're on the same step.

terliuk, 2021-06-04:
> This one points directly to epix. We discussed this in the MC meeting where the splines propagation was presented and (I think) agreeed on not duplicating to much epix does already, right? Meaning [...]
Yes, it will be direct passage from epix during instruction generation and will only be used for generation of scintillation times to individual photons and not actual quanta generation.

terliuk, 2021-07-02:
I think we can close this issue as resolved, since NEST part was merged to WFsim, right?

closed 2021-07-02 by terliuk