XENONnT/WFSim #170: Split and refactor core.py

terliuk, 2021-06-14:
Our core.py is unreasonably long and very hard to navigate. Does it make sense to split it into separate files corresponding to pulse class, S1, S2 etc? I think it will hugely improve development and testing of our MC.

petergaemers, 2021-06-15:
Since it's not a lot of work it's fine by me if it helps people

zhut19, 2021-06-15:
We might want to follow [this](https://devblogs.microsoft.com/oldnewthing/20190916-00/?p=102892) blog?

ramirezdiego, 2021-07-12:
Are we rescuing this?

zhut19, 2021-07-12:
We should test this on some other repo first. And since it's not urgent right now, I will put it on hold unless someone else would like to do it.

terliuk, 2021-07-13:
I would say that advantages of split core is actually more important than having git blame. In the worst case, people can trivially "go" back in time to commits/previous versions and use blame there.

PR #188 (2021-07-14): Core splitting 2
**What is the problem / what does the code in this PR do**
Another attempt of #181 to fix #170
closed 2021-07-14 by zhut19