XENONnT/WFSim #174: Match straxen config with fax_config in certain parameters.

ricmperes, 2021-06-18:
Currently, set fax_config options are not propagated to straxen and event processing. While it might be useful to a certain extent, for some parameters such as the `drift_velocity_liquid` (`electron_drift_velocity` in straxen), it leads to a general misreconstruction of events.

- Example: mono-energetic ER 30 kev

Not setting the straxen parameter manually. |  After changing straxen context config with electron_drift_velocity=("electron_drift_velocity_constant", v_drift, False)
----------- | ------------
[image] | [image]

ramirezdiego, 2021-06-18:
Hey Ricardo, thank you for looking so close and for opening the issue. Notice that the related general problem has already been outlined (e.g., here: https://github.com/XENONnT/straxen/issues/478) and we are working on a centralized MC context solution that will be available in the next days. The problem, as usual, is that things in the MC side evolve slowly and don't catch the attention they probably deserve.

The new context will enable a flexible query of the CMT options for event processing. But, in case CMT is not requested, this will default to the fax_config definitions. It is therefore important that we (cc @zhut19, @petergaemers, @terliuk, ...) make sure that these manually set options are correctly propagated. The idea proposed by @WenzDaniel would enable the option of feeding the custom configs as a dictionary to the context. But this would still require for the user to fetch the configs from the fax_cofig file and generate this dictionary. This can be done in the mc_chain scripts automatically, but someone running WFSim in a notebook risks to overlook this part... 🤔 @WenzDaniel et al., should we re-discuss this part?

And sorry this is still a problem... Long battle :)

WenzDaniel, 2021-06-18:
As a said we can put a context in cutax which replaces CMT settings with hardcoded values from the fax_config file. However, I do not understand how the detector parameters can be "wrong". In the end you would like to match MC and real data so I guess you should use in your MC the same values as stored in CMT and not vise versa is not it?

ramirezdiego, 2021-06-18:
If the question is for me, I didn't get it.

zhut19, 2021-09-13:
Fixed with https://github.com/XENONnT/straxen/pull/602

closed 2021-09-13 by zhut19