XENONnT/WFSim #182: S2 reconstruction bias when using noise (enable_noise)

dantonmartin, 2021-07-06:
The S2 reconstruction bias seems to be altered by the addition of noise. When using 

`st.set_config(dict(fax_config_override=dict( s2_luminescence_model='simple',enable_noise=1,)))`

, the bias for Ar37-like S2s is in the order of 7%. When enable_noise=0, it is (within statistics) in the order of 0%.
I then re-ran the simulation with slightly higher energies (31.5 keV for Kr83m-like events (no double-S2 though)) and the same effect shows up, although with the bias magnitude being reduced to 2-3%. This points at a constant offset in the area reconstruction, like a wrong baseline. A couple plots showing the effect for Ar37-like.

[image]
[image]

I will keep digging and update if I find anything.

skazama, 2021-07-07:
Thanks @dantonmartin for your careful checks! Maybe I'm wrong, but we also see similar bias in @jorana's study (See[ this fig](https://xe1t-wiki.lngs.infn.it/lib/exe/fetch.php?media=xenon:j_angevaare:rec_bias_compare_lowe_peak_split_gof_th-none0_s1_max_rise_time_-200_s2_merge_max_dura-50000_s1_max_rise_time-110_s2_merge_max_gap-10000.png)). I guess @zhut19 has some ideas to resolve this issue

HenningSE, 2021-07-07:
Stupid question: The noise file has a mean value of 0?

dantonmartin, 2021-07-07:
As far as I understand, it doesn't really matter if there is a (small)
offset from 0 as long as the offset is constant, since then strax will
account for the offset during baseline estimation. I will check but Peter G
seems to have found a strange bug regarding baseline construction which
could be the source of the problem.

zhut19, 2021-09-13:
It turns out to be not a wfsim issue. Fixed with https://github.com/XENONnT/straxen/pull/601

closed 2021-09-13 by zhut19