XENONnT/WFSim #192: nVeto pulses haven't made following Geant4 file

mzks, 2021-07-25:
I found among timing check that WFSim doesn't make nVeto pulse following Geant4 file.
1) Events which have many hits in Geant4 file have not been seen in produced `truth_nv` or `records_nv`.
2) Events which have no nVeto hits in G4, have `records_nv` and `truth_nv`

I guess that nVeto events are shifted in somewhere...
https://github.com/XENONnT/WFSim/blob/018694ba105743b4dcc4b01346052a05f579d8d3/wfsim/strax_interface.py#L134
At this line, G4 array is flatten. There may be a bug in chopping the events after that line. 
We need to check them carefully.

Rest information I investigated is there.

I used this file and tried to make pulses with very low rate.
/dali/lgrandi/xenonnt/simulations/testing_epix_wfsim/tpc_and_nveto_cryoneutrons_200.root
python	v3.7.9	(default, Dec 15 2020, 15:48:53) [Clang 11.0.0 ]
strax	v0.16.0	/Users/mzks/.pyenv/versions/3.7.9/lib/python3.7/site-packages/strax
straxen	v0.19.2	/Users/mzks/xenon/straxen/straxen
wfsim version:0.5.6
epix version:0.1.1

Scripts: 
```
import strax, straxen, wfsim, epix, nestpy
import numpy as np
import pandas as pd
import matplotlib.pyplot as plt
import uproot

st = straxen.contexts.xenonnt_simulation()
st.register(wfsim.RawRecordsFromMcChain)
epix_config = {
    'cut_by_eventid': True,         # means that entry_start and entry_stop refer to the original G4 event id
    'debug': True,                  # print runtime outputs
    'source_rate': 0,               # do not alter G4 time of events, because this will be done in WFSim
    'micro_separation_time': 10.,   # ns; default epix separation for clustering decision
    'max_delay': 1e7                # ns; default epix cut for long events (time wrt to first interaction in TPC)
}
epix_config['detector_config_override'] = None
epix_config['micro_separation'] = 0.05
epix_config['tag_cluster_by'] = 'time'

st.set_config(dict(
    detector='XENONnT',
    #fax_file='/Users/mzks/xenon/mc/mc/build/bin/laser_test10_10000.root',
    fax_file='/Users/mzks/xenon/WFSim/bench/tpc_and_nveto_cryoneutrons_200.root',
    event_rate=1.,
    chunk_size=5.,
    entry_start=0,
    #entry_stop=20,
    #fax_config='fax_config_nt_sr0_v0.json',
    fax_config='fax_config_nt_design.json',
    fax_config_override=dict(url_base='https://raw.githubusercontent.com/XENONnT/private_nt_aux_files/master/sim_files',
                             enable_electron_afterpulses=False),
    epix_config=epix_config,
    targets=('tpc',))
)
st.set_config(dict(
    neutron_veto=True,
    fax_config_nveto='fax_config_nt_nveto.json',
    fax_config_override_nveto=dict(enable_noise=False,
                                   enable_pmt_afterpulses=False,
                                   enable_electron_afterpulses=False,
                                   photon_area_distribution='/Users/mzks/xenon/WFSim/bench/private_nt_aux_files/sim_files/XENONnT_spe_distributions_nveto_20210722.txt',
                                   pmt_transit_time_mean=46.0),
    targets=('tpc', 'nveto',),
))

run_id = 'timing_check_01'
rr = st.get_array(run_id, 'raw_records', progress_bar=False)
truth = st.get_array(run_id, 'truth', progress_bar=False)
rr_nv = st.get_array(run_id, 'raw_records_nv', progress_bar=False)
truth_nv = st.get_array(run_id, 'truth_nv', progress_bar=False)

```

truth['g4id'] is 
```
array([ 11,  11,  11,  11,  11,  11,  11,  11,  11,  11,  11,  11,  11,
        11,  11,  11,  11,  11,  12,  12,  12,  13,  13,  16,  16,  16,
        16,  16,  16,  16,  16,  16,  18,  18,  18,  18,  18,  21,  21,
        21,  21,  23,  23,  23,  23,  23,  23,  23,  23,  23,  23,  23,
        23,  23,  24,  24,  24,  25,  25,  25,  25,  25,  25,  25,  25,
        25,  25,  25,  26,  26,  26,  31,  31,  36,  38,  38,  38,  38,
        48,  48,  48,  48,  48,  48,  48,  48,  48,  48,  48,  48,  48,
        50,  50,  50,  50,  53,  53,  53,  53,  53,  53,  53,  53,  53,
        54,  55,  55,  55,  55,  55,  55,  55,  55,  55,  55,  55,  55,
        55,  55,  55,  55,  55,  56,  56,  56,  60,  60,  62,  62,  62,
        62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,
        62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,
        62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,  62,
        62,  62,  62,  62,  62,  62,  62,  62,  62,  64,  64,  64,  64,
        64,  64,  69,  69,  69,  69,  69,  69,  74,  74,  76,  76,  76,
        76,  76,  76,  76,  76,  76,  76,  76,  76,  76,  77,  77,  77,
        77,  77,  77,  77,  77,  77,  77,  77,  77,  77,  77,  77,  77,
        77,  77,  77,  77,  77,  77,  77,  77,  77,  77,  77,  77,  77,
        77,  77,  77,  77,  77,  77,  77,  77,  77,  82,  82,  82,  82,
        82,  82,  87,  87,  90,  90,  90,  90,  91,  91,  96,  96,  96,
        96,  96,  96,  96,  96,  96,  96,  96,  96,  96, 105, 105, 105,
       105, 105, 105, 105, 105, 105, 105, 105, 105, 105, 105, 105, 105,
       105, 106, 106, 106, 106, 106, 106, 106, 107, 107, 107, 107, 107,
       108, 108, 108, 108, 108, 108, 108, 108, 108, 108, 108, 108, 108,
       108, 108, 108, 108, 108, 109, 109, 109, 109, 109, 109, 109, 109,
       109, 109, 114, 114, 114, 114, 114, 114, 114, 122, 122, 122, 122,
       122, 122, 122, 122, 129, 130, 130, 130, 130, 130, 130, 130, 130,
       130, 130, 130, 130, 130, 130, 130, 130, 130, 130, 130, 130, 130,
       130, 130, 132, 132, 132, 132, 132, 132, 132, 132, 132, 133, 133,
       133, 133, 135, 135, 135, 136, 136, 136, 136, 136, 136, 136, 136,
       136, 136, 136, 138, 138, 140, 140, 140, 140, 140, 140, 140, 140,
       144, 144, 144, 144, 147, 147, 147, 147, 147, 147, 147, 150, 150,
       150, 151, 151, 151, 153, 153, 153, 153, 153, 153, 153, 153, 154,
       154, 157, 157, 157, 157, 159, 159, 164, 164, 166, 166, 166, 166,
       166, 166, 166, 166, 166, 166, 166, 166, 166, 172, 172, 172, 172,
       173, 173, 173, 173, 173, 179, 179, 179, 182, 182, 182, 186, 186,
       186, 186, 186, 186, 186, 186, 186, 186, 186, 186, 186, 186, 186,
       186, 186, 186, 186, 186, 186, 186, 186, 186, 189, 189, 189, 189,
       189, 189, 190, 190, 190, 190, 190, 190, 191, 194, 194, 194, 194,
       197, 197, 197, 197, 197, 197, 197, 197, 197, 197, 197, 197, 197,
       197, 197, 197, 199, 199, 199, 199, 199, 199, 199], dtype=int32)
```
Example: 

In Original Geant4 file, event 0, `g4_pmthitID[0]` have many available IDs for nVeto.
`[2037, 2051, 2045, ... 2090, 2028, 2114]`

In Original Geant4 file, event 11, `g4_pmthitID[11]` have no available ID for nVeto.
`[106, 137, 314, 255, ... 276, 489, 438]` 
All of them are not nVeto ID (2000--2199)

ramirezdiego, 2021-07-26:
Hi Keita, I am not sure I understand your example, could you please be a bit more clear?
I see that `tpc_and_nveto_cryoneutrons_200.root` has, for `eventid==11`, hits in TPC PMTs. This makes me worry and I would say this is the first think to check on the G4 MC, to discard bugs. You mean that this event has `records_nv`? So these TPC PMTs are treated as nVeto PMTs?

For the rest, I do not understand the problem. For all the other `eventid` with nVeto hits, do you see the information after WFSim?

I would start by showing an array of `eventid` in which there are `pmthitID` in the nVeto range. Then, check if all these ids are afterwards present in the `truth_nv`. Could you do that?

petergaemers, 2021-07-26:
Hi Keita, what does "some records_nv have been observed mean?

mzks, 2021-07-26:
I will show the case g4 ID 11 as an example.

An event 11 in Geant file don't have no nVeto hit but only TPC 
[image]
As you know, nVeto IDs start in 2000. There are no nVeto ID in this event.
They are TPC ID, but it is not directly problem. We can discard the wrong ID in `_read_optical_nveto()` in WFSim.

However, WFSim generates `raw_records_nv` for the event 11. I found the `raw_records_nv` around the `truth_nv['time']`
(`_rr_nv` is the selected `raw_records_nv` as in Event 11).
[image]

For nVeto, WFSim simply makes pulse when it find the photon on the PMT in principle but the IDs are not match. That's the issue.

petergaemers, 2021-07-26:
Got it. What do you propose as a solution?

ramirezdiego, 2021-07-26:
@petergaemers Didn't we have a cut on the nVeto channels, for this case? I vaguely remember we discussed, but I can't find it from a first glance.
Namely, for the read_optical to be called from the McChain class only considering PMT id's in the nVeto range.

mzks, 2021-07-26:
If I would have a proposal, I posted this in Pull-request.

terliuk, 2021-07-26:
One small thing, is it only me or photons are sorted by time? i remember that we had issues due to implicit sorting in PMT hit time generation, which was not supposed to be there.

mzks, 2021-07-28:
/home/mzks/chain_test/nveto_g4_bug_fix1.ipynb Note on dali

mzks, 2021-07-28:
https://xe1t-wiki.lngs.infn.it/doku.php?id=xenon:xenonnt:mc:telecons:210728:nveto_bug_note HTML style

mzks, 2021-08-04:
Solved. https://xe1t-wiki.lngs.infn.it/doku.php?id=xenon:xenonnt:dsg:nveto:calibration_design:wfsim_laser_trial1

mzks, 2021-08-04:
But before close this issue, I would like to do more careful check. Thank you very much @petergaemers