XENONnT/WFSim #287: S1/S2 grouping in truth

PavelKavrigin, 2021-12-24:
Currently the information written to the truth output of WFSim is reduced: 

All S1s within 100 ns separation are merged:
https://github.com/XENONnT/WFSim/blob/0d74d97f1e534f0383fdd2791eca5be8e26831b0/wfsim/core/rawdata.py#L109

All S2s within 2 mm separation are merged:
https://github.com/XENONnT/WFSim/blob/0d74d97f1e534f0383fdd2791eca5be8e26831b0/wfsim/core/rawdata.py#L114

As a result, the truth cannot be used for any analysis which requires a knowledge of true multi-scatters in the instructions since this information is lost if the interactions happen close to each other. The most simple fix would be just to change these thresholds, e.g. 100 ns to 0.1 ns, 2 mm to 1 um. In case if we want to keep a possibility to reduce the size of the truth output, we will need to add some flag to the configuration to specify whether we want to merge instructions in the truth or not.

JoranAngevaare, 2021-12-28:
Seems to be part of https://github.com/XENONnT/WFSim/pull/125, specifically this commit: https://github.com/XENONnT/WFSim/pull/125/commits/8de52e04b56ad204f4e92d37432d53b4031c771e

In my view, your proposal makes a lot of sense

zhut19, 2022-01-11:
Fixed in #288

closed 2022-01-11 by zhut19