XENONnT/WFSim #420: NaNs in t_last_photon

JYangQi00, 2023-09-15:
The `truth` table generated by WFSim occasionally produces NaN values in t_last_photon despite the fact that the instructions seem to indicate that the event should be detectable. The following event is an example of this:

```
event_number,type,time,x,y,z,amp,recoil,e_dep,g4id,vol_id,local_field,n_excitons,x_pri,y_pri,z_pri
140,1,1646832231842000000,-64.71233,9.902024,-133.29727,143,7,3.4371514,0,0,22.369154340346643,6,0.0,0.0,0.0
140,2,1646832231842000000,-64.71233,9.902024,-133.29727,103,7,3.4371514,0,0,22.369154340346643,6,0.0,0.0,0.0
```

ramirezdiego, 2023-09-21:
@JYangQi00 Did you verify that this only happens for interactions beyond the sensitive region? Can the issue be closed?

JYangQi00, 2023-09-25:
My apologies for the late reply. I did verify that this issue only happens to S2s in the charge insensitive volume (CIV). However, there also seems to be some points in the CIV which do not get a NaN photon times in the truth table, when perhaps they should?

[image]